WifWaf/TCA9548A #4: Multiple OLED devices with ESP32

thijstriemstra, 2020-07-21:
I'm trying to use multiple 0.91 inch OLED displays (128x32px) with an ESP32 board and a TCA9548A. I found this: https://gist.github.com/johnaboxall/653cc12aa66571b6490610c20ac3b7c8 and it seems like a good start but I'd rather use a library like this. Can you show how to port that example using this library?

WifWaf, 2020-07-21:
[Something like this](https://pastebin.com/8tnW7jEj)?

I can't test it, but it should give you an idea.

thijstriemstra, 2020-07-21:
thanks I'll give it a try asap and report back.

thijstriemstra, 2020-08-11:
I tested with 2 1.3" SH1106 128x64 OLEDs and this library: https://github.com/ThingPulse/esp8266-oled-ssd1306

```cpp
#include <Arduino.h>
#include <Wire.h>
#include <TCA9548A.h>
#include <SH1106Wire.h>

int displayWidth = 128;
int displayHeight = 64;

TCA9548A I2CMux(0x70); 
TwoWire twire = TwoWire(1);
SH1106Wire display(OLEDAddress, I2CMuxSDAPin, I2CMuxSCLPin, GEOMETRY_128_64, I2C_TWO);

void setup() {
  Serial.begin(115200);

  I2CMux.begin(twire, I2CMuxSDAPin, I2CMuxSCLPin);
  I2CMux.closeAll();

  // draw on screen 1
  I2CMux.openChannel(0);
  display.init();
  display.flipScreenVertically();
  display.clear();
  display.setFont(ArialMT_Plain_24);
  display.setTextAlignment(TEXT_ALIGN_CENTER);
  display.drawRect(0, 0, displayWidth, displayHeight);
  display.drawString(64, 20, "0");
  display.display();
  I2CMux.closeChannel(0);

  // draw on screen 2
  I2CMux.openChannel(1);
  display.init();
  display.flipScreenVertically();
  display.clear();
  display.setFont(ArialMT_Plain_24);
  display.setTextAlignment(TEXT_ALIGN_CENTER);
  display.drawRect(0, 0, displayWidth, displayHeight);
  display.drawString(64, 20, "1");
  display.display();
  I2CMux.closeAll();

  Serial.println("Started.");
}

void loop() {
}
```

It needs the fix for #5 though. Seems to work!

closed 2020-09-04 by thijstriemstra