WifWaf/TCA9548A #5: Cannot assign different SCL/SDA pins

thijstriemstra, 2020-08-11:
the `begin()` method accepts a TwoWire instance and then calls `begin()` on that instance:

https://github.com/WifWaf/TCA9548A/blob/master/src/TCA9548A.cpp#L15

Instead it should call `twowire.begin(sda, scl)`. Or this library should not call `begin()` on the instance and let the user do this (so the custom sda/scl pins can be passed then).

I made this change to `begin` and now it works:

```cpp
void TCA9548A::begin(TwoWire &inWire, int sda_pin, int scl_pin)
{
    this->myWire = &inWire;
    this->myWire->begin(sda_pin, scl_pin);
}
```

WifWaf, 2020-08-11:
It was removed here 5fbbedbe326e2c2a5096e7ac49634df240389b16

If I remember correctly... the arguments were removed as the UNO will not compile. It has a single hardware I2C port and doesn't have an overload function for specifying which pins you want to use. 

I think new Arduino users tend to get confused/forget and in sketches and so it causes problems/issues to be raised.

Not sure on your device, however, the[ESP32 can take a setpin argument](https://github.com/espressif/arduino-esp32/blob/4d98cea085d619bed7026b37071bd8402a485d95/libraries/Wire/src/Wire.cpp#L61) which should allow this to work without changing the begin function.

Perhaps adding preprocessor directives to switch the function out for the UNO would be the best solution (unless there's something more obvious).

thijstriemstra, 2020-08-11:
> If I remember correctly... the arguments were removed as the UNO will not compile.

Ok, if there's an exception for the Uno I agree "adding preprocessor directives" sounds like a good thing to do. But I haven't encountered any differences, or compile errors between Uno/Nano/ESP32 using TwoWire.

thijstriemstra, 2020-08-11:
> Or this library should not call begin() on the instance and let the user do this (so the custom sda/scl pins can be passed then).

This is also an option of course. Users would have to call `begin` on their TwoWire instance themselves.

WifWaf, 2020-08-11:
[See here for the UNO begin functions](https://github.com/arduino/ArduinoCore-avr/blob/master/libraries/Wire/src/Wire.cpp).

From what I can see, any library using the official ArduinoCore will not compile with a begin(sda, sdl) type function - I've double-checked this on a UNO and MEGA for sanity.

> This is also an option of course. Users would have to call begin on their TwoWire instance themselves.

I think this would cause confusion as it's not convenient as it's usually always unnecessary. To be honest, I suspect this is why the ESP32 has a setpin function. 

It might actually be best to include a commented in the example to show the workaround setpin function.

thijstriemstra, 2020-08-11:
> I think this would cause confusion as it's not convention as it's usually always unnecessary. 

Not sure, when using TwoWire I expect to do some extra work myself, so I wouldn't mind.

>  It might actually be best to include a comment in the example to show the workaround setpin function. *excuse the spelling errors

Looking forward to a workaround.

WifWaf, 2020-08-11:
Yea - I get you.

The setPins is recent (4d98cea085d619bed7026b37071bd8402a485d95), so yes, work with your fork. 

I'll add this to relevant repository examples once it's in the stable arduino-esp32 branch.

I'll keep this issue open until then.

thijstriemstra, 2020-08-12:
Ah nice find, that's a good addition to arduino-esp32.

thijstriemstra, 2020-10-03:
@WifWaf I just upgraded to Espressif 32 2.0, does this mean `setPins`'s available now?

WifWaf, 2020-10-03:
It wasn't included in the [stable 1.0.4 release yesterday](https://github.com/espressif/arduino-esp32/releases/tag/1.0.4) (I think this is 2.0 on PIO) .. for some reason.

It is still in the master branch and looks like [Adafruit are waiting ](https://github.com/espressif/arduino-esp32/issues/3779) for it as well - so hopefully in the next one?

WifWaf, 2020-11-03:
[It's been included in today's pre-release](https://github.com/espressif/arduino-esp32/blob/be4d3b6cb87236a749a0abc8288b49cb6b79710b/libraries/Wire/src/Wire.cpp#L61). Once it's out of pre-release and PIO update their version - I'll add the changes.

Good times.

WifWaf, 2020-12-27:
setPins has been added to the examples due to the recent update - it should be usable on PIO very soon.

thijstriemstra, 2020-12-27:
> to the recent update 

What specific version?

WifWaf, 2020-12-27:
I was going to say 1.0.5, but [checking the json file I'm using, it seems it's still 1.0.4](https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json). 

Would you mind checking if your Arduino IDE compiles Wire.setPins(sda, scl)?

Seems I might an issue with my Arduino IDE...

thijstriemstra, 2020-12-27:
I don't use Arduino IDE, I'll give it a try with PlatformIO.

thijstriemstra, 2020-12-27:
Not yet... 

[image]

Using espressif 32 2.1.0

WifWaf, 2020-12-27:
I guess I installed a development version previously which is causing issues. How ironic, I only use Arduino IDE for testing library compatibility.

I'll wait on PIO for versioning in future - cheers for the verification.

thijstriemstra, 2021-05-17:
Looks like it!

```c++
public:
    TwoWire(uint8_t bus_num);
    ~TwoWire();
    
    //call setPins() first, so that begin() can be called without arguments from libraries
    bool setPins(int sda, int scl);
```

closed 2021-05-17 by thijstriemstra