WifWaf/TCA9548A #6: I2C bus scanner example

thijstriemstra, 2020-08-11:
Would be cool to have an example that scans each bus for devices in a slow loop. I made this but it doesn't show the bus nr (and its hardcoded in my copy of the library):

```cpp
void TCA9548A::scan() {
  byte error, address;
  int nDevices;
  Serial.println("Scanning...");
  nDevices = 0;
  for (address = 1; address < 127; address++ ) {
    this->myWire->beginTransmission(address);
    error = this->myWire->endTransmission();
    if (error == 0) {
      Serial.print("I2C device found at address 0x");
      if (address<16) {
        Serial.print("0");
      }
      Serial.println(address,HEX);
      nDevices++;
    }
    else if (error==4) {
      Serial.print("Unknow error at address 0x");
      if (address<16) {
        Serial.print("0");
      }
      Serial.println(address,HEX);
    }    
  }
  if (nDevices == 0) {
    Serial.println("No I2C devices found\n");
  }
  else {
    Serial.println("done\n");
  }
  delay(5000);          
}
```

WifWaf, 2020-08-11:
Sure - if you can create a pull request with the nested loop / I2C scanner example I'll add it to the examples folder.

thijstriemstra, 2020-08-11:
Sounds good.

closed 2020-08-11 by WifWaf