WifWaf/TCA9548A #8: Compile warnings

thijstriemstra, 2020-09-04:
```
.pio/libdeps/esp-wrover-kit/TCA9548A/src/TCA9548A.cpp: In member function 'uint8_t TCA9548A::read()':

.pio/libdeps/esp-wrover-kit/TCA9548A/src/TCA9548A.cpp:84:54: warning: ISO C++ says that these are ambiguous, even though the worst conversion for the first is better than the worst conversion for the second:

     this->myWire->requestFrom(this->_address, 1, true);

 ```

Would be nice to have a clean build.

WifWaf, 2020-09-05:
Ah, the Wire.cpp has a bunch of overload functions.

Try pulling the latest changes, it should remove the ambiguity.

thijstriemstra, 2020-09-05:
thanks, looks good now. Can you release this?

WifWaf, 2020-09-05:
Sure - it might take a while to appear in any library managers.

closed 2020-09-08 by thijstriemstra