WifWaf/TCA9548A #9: Error including the library with Arduino Mega

cmullica, 2021-02-04:
Hi, incredibly new at this, but I'm having some issues even incorporating your library at all. When taking a blank script (setup and loop remaining empty) such that:

#include <TCA9548A.h>

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}

is the entirety of the script. 
It's yielding an error upon verifying:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\src/Wire.h:66:13: note: candidate: uint8_t TwoWire::requestFrom(uint8_t, uint8_t, uint8_t)
     uint8_t requestFrom(uint8_t, uint8_t, uint8_t);
             ^~~~~~~~~~~
C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\src/Wire.h:69:13: note: candidate: uint8_t TwoWire::requestFrom(int, int, int)
     uint8_t requestFrom(int, int, int);
             ^~~~~~~~~~~
exit status 1
Error compiling for board Arduino Mega or Mega 2560.

Any ideas as to what's going on or how I might rectify this?

WifWaf, 2021-02-04:
No problem!

The Arduino core libraries have slight differences between MCUs, and so the libraries do as well.

The compiler is not happy about an "ambiguity" between which function variant is being called.

I've forced it to pick one (hopefully won't break any other MCUs!).

If you update your files with the change,  it should work now.

Let me know if you have any other issues!

cmullica, 2021-02-05:
Thanks for the swift response!
Quick question: I can see that the .cpp has been updated, but the .h has not. 
Doesn't the .h need to be updated in order to be implemented or should I uninstall and reinstall the library?

closed 2021-02-05 by cmullica