WifWaf/TCA9548A #12: TCA9548 in multi master mode

Manhem, 2024-09-20:
Hi.
Can your library work in multi master mode?
I have a software where I use your library to talk to different I2C buses.
One of this bus, there is also connected another equipment that is also talking to the device on that I2C bus.
This equipment I do not have any control over.
Can your library detect if the bus is already occupied when I turn the bus on?
If it finds it occupied, how can I track this?
It happend very offen that both are talking at the same time and when this is happend everything stops working.

WifWaf, 2024-09-23:
Hey, 

This is something which is usually implemented at the project level, rather than library.  This library doesn't do anything fancy and it uses your TwoWire instance.

I've never personally used a multi-master setup, however, I imagine you'd need access to the firmware for both devices to stop interference ideally.

I would try identifying the other master device's transmission timing, then you can transmit in between. 

You could try setting the SCL line pin to an input and attach an interrupt to detect the start of a transmission (pulled low). Then you can work out the timings and switch the SCL pin from a generic input to I2C sCL when you need to make a transmission.

Might be worth trying.

Manhem, 2024-09-25:
Do you have any code that you are using that you can share to me?
Maybe something from other projects you have done?